eclipse-tahu/tahu #11: node-red-contrib-sparkplug still references old repostitory

ElectronHacks, 2019-01-16:
flows.nodered.org and npmjs.com both reference the old Cirrus-Link Sparkplug repository. 
I have numerous issues with Sparkplug inside of Node-RED and am not sure where to report my bugs, is this the place? Who can fix the npmjs repository?
I imagine the issues should be submitted individually but wanted to attach them so you know what they are.

1. Node-red-contrib-sparkplug cannot work without a MQTT user and password.
If you drags the SparkPlug node onto a flow and click Deploy you will get two errors:
	1. "Error creating new client"
	2. "TypeError: Cannot read property 'on' of undefined"
Then edit the SparkPlug node and type in a user name and password then click Deploy and you the errors will be gone. You can make it do it again by deleting out user and password.

2. Sparkplug MQTT Keepalive pings is hard coded every 5 seconds which should be configurable
Pinging every 5 seconds is ok if bandwidth is of no concern and you really care about statefulness. On battery power or small data plan this time should be adjustable. I have been working on how to add this to the Node-Red sparkplug node configuration but I haven't gotten it working yet.

3. Shouldn't we be able to skip sending the "Type" with every value?
If you try to send DDATA with only the name and value, the node errors with:
"TypeError: Cannot read property 'toUpperCase' of undefined" so it can't handle sending data without the data type. This seems wasteful with bandwidth to have to always send the datatype every single time, cant it just be sent at birth? I did spend some time reading the specification and can't see where its required, I may be wrong though.

jtmoderate876, 2019-01-17:
Same here with 

> If you drags the SparkPlug node onto a flow and click Deploy you will get two errors:
> 
> "Error creating new client"
> "TypeError: Cannot read property 'on' of undefined"
> Then edit the SparkPlug node and type in a user name and password then click Deploy and you the errors will be gone. You can make it do it again by deleting out user and password.
>

theonlypwner, 2021-02-27:
This occurs because of

https://github.com/eclipse/tahu/blob/v0.5.9/client_libraries/javascript/node-red-contrib-sparkplug/sparkplug/sparkplug.js#L86-L94

https://github.com/eclipse/tahu/blob/v0.5.9/client_libraries/javascript/sparkplug-client/index.js#L25-L47

The first error occurs because an exception is thrown and caught when `username` is missing. The call to `node.error` is incorrect; it should contain one exception object as the first argument, and the original `msg` as the second argument (which should be omitted or `undefined` in this case).

This means we get one error `"Error creating new client"` with the original exception swallowed. The catch block doesn't prevent the following code from running, so `sparkplugClient.on('birth', ...);` results in `"TypeError: Cannot read property 'on' of undefined"` because `sparkplugClient` is `null` after the constructor threw.

beckerben, 2021-04-22:
I can confirm this bug still exists, bypassed it by setting up broker with authentication

guillaumeklimek, 2021-08-19:
I can confirm this bug still exists.