eclipse-tahu/tahu #33: Subscribing to an existing tag in Ignition

mleefers1729, 2019-09-27:
I see in the on_connect function we are subscribing to some topics. How do we actually subscribe to a specific tag in Ignition and start listening to that tag value? I've tried a few different things and I guess I cant seem to get the path right. It doesn't seem all that intuitive?

TMChenier, 2019-09-28:
You do not make subscriptions for tags. The subscription is for the node or device that is publishing the metric(tag), then parse through the payload to find the specific metric you're interested in. 

I'm assuming you have Ignition setup with the MQTT Transmission module to publish data?

mleefers1729, 2019-09-28:
Ohh okay that makes more sense. Yeah I have an Edge gateway transmitting to a main gateway with the transmission/engine running so I know that the server is publishing data.

closed 2019-09-30 by mleefers1729