eclipse-tahu/tahu #34: Negative integer measurements raise an error when published

atodorski, 2019-10-03:
If I define a measurement with an Int32 (or 8, 16, or 64) then attempt to publish a negative value I get the following error:

...
    metric = addMetric(payload, None, AliasMap.SOME_VALUE, MetricDataType.Int32, int(SOME_VALUE))
  File "/......./sparkplug_b.py", line 182, in addMetric
    metric.int_value = value
ValueError: Value out of range: -4566

TMChenier, 2019-10-11:
I've also run into this issue. The problem is Sparkplug wants to store any int value metric (signed or unsigned) as an unsigned value in the protobuf message (see page 44 of the spec). The standard python Google protobuf libraries do range checks on values as the message is being built. So trying to store a negative number as an unsigned int type will fail.

I'm not sure if the protobuf libraries in other languages do these same checks and will just cast the variable as such. 

Unfortunately, I haven't had the time to figure out a solution. Possibly conditioning a negative value in some way before building the metric?

atodorski, 2019-10-14:
I don't recall seeing this issue using the java client, so this seems like
more of a problem with protocol buffers to me if it behaves differently for
different languages (kind of defeats the purpose).

-- 
Adam Todorski
Senior Technical Solutions Architect
518-669-6778
adam.todorski@auto-grid.com

-- 
CONFIDENTIALITY NOTICE:
The information contained in this e-mail and any 
files transmitted with it is confidential and intended only for the review 
and use of the recipient(s) named above. If the reader of this message is 
not the intended recipient, you are hereby notified that you have received 
this email in error, and that any review, dissemination, use or copying of 
this email or any attachment hereto is strictly prohibited. If you have 
received this email in error, please delete it and all copies from your 
system and notify of the error by return e-mail to the sender.

commit ca2d4164 (2022-06-21): Fixes #34 by converting negative values as needed before publishing
commit db8f56b6 (2022-06-21): Merge pull request #203 from wes-johnson/34

Fixes #34 by converting negative values as needed before publishing
closed 2022-07-20 by wes-johnson