eclipse-tahu/tahu #39: node-red-contrib-sparkplug, not possible to send DDEATH

tord-v, 2019-11-22:
# Problem
The direct result of that change is that devices cannot get individual DDEATH messages. Which i think is quite important when one Node can hold several devices. QOS data will be kept "Good" :(

### I am sending the following message into the Sparkplug node:
`
msg={ "topic" : "Equipment2/DDEATH",'payload': {"timestamp" : Date.now()} };
`

I get this code back:
`
"ReferenceError: options is not defined"
`
### I think i found a bug in the code
I think the problem is related to a missing parameter in the code:
node-red-contrib-sparkplug/sparkplug/sparkplug.js, calls publishDeviceDeath:
```
else if (messageType === "DDEATH") { 
// Clear device cache 
delete deviceCache[deviceId]; `
// Publish device data sparkplug
Client.publishDeviceDeath(deviceId, payload);
```
which javascript/sparkplug-client/index.js expect the option parameter, which is not included.

```
this.publishDeviceDeath = function(deviceId, payload, **options**) { var topic = version + "/" + groupId + "/DDEATH/" + edgeNode + "/" + deviceId;
// Add seq number
addSeqNumber(payload); 
logger.info("Publishing DDEATH for device " + deviceId); 
client.publish(topic, encodePayload(maybeCompressPayload(payload, options))); 
messageAlert("published", topic, payload);
```
Crossing my fingers for some feedback or even a fix :)
Keep up the good work!

tord-v, 2019-11-23:
Just to ble clear, the bug is fixed this way:
in
client_libraries/javascript/sparkplug-client/index.js
line 268 as is:
`
client.publish(topic, encodePayload(maybeCompressPayload(payload, options)));
`
Should be:
`
client.publish(topic, encodePayload(maybeCompressPayload(payload, {})));
`
As the function does not have this variable (as is code):
```
this.publishDeviceDeath = function(deviceId, payload) {
        var topic = version + "/" + groupId + "/DDEATH/" + edgeNode + "/" + deviceId;
        // Add seq number
        addSeqNumber(payload);
        // Publish
        logger.info("Publishing DDEATH for device " + deviceId);
        client.publish(topic, encodePayload(maybeCompressPayload(payload, options)));
        messageAlert("published", topic, payload);
    };
```

ckienle, 2019-11-26:
Thanks for reporting this, I've committed a fix 05ad8ed6e67b062b343ea606469beb8e1802f378

closed 2019-11-26 by ckienle

tord-v, 2019-12-17:
Has the commit been removed @ckienle ? it was working before, but i cannot se it anymore in the main repo...