eclipse-tahu/tahu #56: sparkplug_b_c_sharp.proto incorrect property name metric (missing the s)

danwale, 2020-03-09:
I'm implementing a C# client and have come across it complaining that there is no property called **metrics** in Payload, on inspection of the .proto file for C# it has incorrectly dropped the **s** from **metrics** and the property is being represented as **metric** instead.

When comparing the two versions of the .proto files it seems like an obvious mistake in the refactoring.

I've found other issues with property dataType instead being datatype but that also appears to be like that in the other .proto file so I'm not sure if that was also a mistake that wasn't used in the java version of the library.

danwale, 2020-03-09:
@wes-johnson I've submitted a pull request #58 that resolves this issue.

wes-johnson, 2020-03-11:
Thanks! This has been approved and merged.

closed 2020-03-11 by wes-johnson