eclipse-tahu/tahu #67: Metric DataSet Column Keys not deallocated

kjcantrell, 2020-05-11:
In free_payload, there is currently no provision for freeing the column keys that are malloc'd as part of init_dataset 
`/*
 * Free memory from an existing Payload
 */
void free_payload(org_eclipse_tahu_protobuf_Payload *payload) {
	int i=0;
	for (i=0; i<payload->metrics_count; i++) {
		free(payload->metrics[i].name);
		// More TODO...
		// JPL 04/05/17... free up string data allocated memory
		if(  payload->metrics[i].which_value == 
			 org_eclipse_tahu_protobuf_Payload_Metric_string_value_tag ) // 15 ???
		{
		  if(payload->metrics[i].value.string_value)  // not null?
		  {
			free(payload->metrics[i].value.string_value);
		  }
		}
		
		if (payload->metrics[i].has_properties)
		{
			int j=0;
			for (j=0; j<payload->metrics[i].properties.keys_count; j++){
				free(payload->metrics[i].properties.keys[j]);
				
				if(payload->metrics[i].properties.values[j].which_value ==
					com_cirruslink_sparkplug_protobuf_Payload_PropertyValue_string_value_tag) 
				{
					if(payload->metrics[i].properties.values[j].value.string_value)
					{	
						free(payload->metrics[i].properties.values[j].value.string_value);
					}
				}
			}
			free(payload->metrics[i].properties.keys);
			free(payload->metrics[i].properties.values);	
		}
	
		
	}
}`

Not sure if this is expected to be freed somewhere else. Potentially not an issue.

jbrzozoski, 2020-05-13:
There are many alloc/free issues with the version currently in tahu.  I was trying to fix all of that in my branch, which is 90% of the way better:
https://github.com/jbrzozoski/tahu/tree/c_client_rework

kjcantrell, 2020-05-13:
> There are many alloc/free issues with the version currently in tahu. I was trying to fix all of that in my branch, which is 90% of the way better:
> https://github.com/jbrzozoski/tahu/tree/c_client_rework

Awesome. Although I think free_payload seems to work well for the most part, where are you seeing your alloc/free issues?

jbrzozoski, 2020-05-15:
For the basic payload fields used in the example, it is fine.  But if the library tries to handle any Sparkplug payloads containing DataSet or PropertySet it can't properly walk through the structures and manage the pointers.

wes-johnson, 2020-08-16:
This code has been reworked thanks to @jbrzozoski. Please try out the latest code on the develop branch.

closed 2020-08-16 by wes-johnson