eclipse-tahu/tahu #79: Free_payload() crashes when having multiple properties - C library

shaila123456789, 2020-06-18:
Hello,

I am trying to explore the C client library of tahu. In this, for publishing device birth message, I have added multiple properties to a metric.
While the DBIRTH message gets published properly, the free_payload() gives exception and the application crashes. The exception is "free(): invalid next size (fast)".

Please guide me how to fix this issue.

I have attached the test file I am using for exploration and a screenshot of exception.

-Thanks,
[test_c.txt](https://github.com/eclipse/tahu/files/4800488/test_c.txt)

S
[image]

jbrzozoski, 2020-06-19:
There are many alloc/free issues with the version currently in tahu. I was trying to fix all of that in my branch, which is 90% of the way better:
https://github.com/jbrzozoski/tahu/tree/c_client_rework
I should really polish up my changes and start a pull request...  It's been a while.

shaila123456789, 2020-06-19:
Hello jbrzozoski,
I took the code from https://github.com/jbrzozoski/tahu/tree/c_client_rework branch. In the test.c file, I added below part in publish_device_birth(), added properties in 2 metrics and added these metrics in dbirth payload.

	org_eclipse_tahu_protobuf_Payload_PropertySet prop = org_eclipse_tahu_protobuf_Payload_PropertySet_init_default;

	add_property_to_set(&prop, "engUnit", PROPERTY_DATA_TYPE_STRING, "milliamps", sizeof("milliamps"));
	add_property_to_set(&prop, "engUnit1", PROPERTY_DATA_TYPE_STRING, "milliamps", sizeof("milliamps"));
	add_property_to_set(&prop, "engUnit2", PROPERTY_DATA_TYPE_STRING, "milliamps", sizeof("milliamps"));
	add_property_to_set(&prop, "engUnit3", PROPERTY_DATA_TYPE_STRING, "milliamps", sizeof("milliamps"));
	add_property_to_set(&prop, "engUnit4", PROPERTY_DATA_TYPE_STRING, "milliamps", sizeof("milliamps"));

	for(uint32_t i = 0; i < 2; i++)
	{
		org_eclipse_tahu_protobuf_Payload_Metric metric1 = org_eclipse_tahu_protobuf_Payload_Metric_init_default;
		init_metric(&metric1, "metric", false, 0, METRIC_DATA_TYPE_INT32, false, false, &i, sizeof(i));

		add_propertyset_to_metric(&metric1, &prop);
		add_metric_to_payload(&dbirth_payload, &metric1);
	}

With this addition, I am getting exception for freeing up the payload :
Encoding succeeded
Free payload memory...
free(): invalid pointer
Aborted

-Thanks,
S

jbrzozoski, 2020-06-19:
As each propertyset is created, it allocates memory, but when you add the propertyset to a metric the pointers within it are copied over, effectively making the metric now the "owner" of those allocations and responsible for freeing them.

In this case, you are copying those pointers twice, and two different metrics think they "own" the pointers, and during the call to free payload, both metrics are trying to free them, causing the second to fail (because they've already been freed).

The proper way to do this is to build two propertyset items and add one to each metric.

shaila123456789, 2020-06-19:
Hello jbrzozoski,

This worked !

-Thanks,
S

shaila123456789, 2020-06-19:
Hello jbrzozoski,

This worked in test.c file.

I compiled this version of c client using make command. Now I am trying to use these libraries and header files in an eclipse C++ project on ubuntu. While compiling the application, I am getting following error:
..//include/tahu/pb.h:825:1: error: template with C linkage
 template <typename GenMessageT> struct MessageDescriptor;

I have included all the header files under extern C.  

Do I need to add any specific macro/flag for this ?

-Thanks,
S

jbrzozoski, 2020-06-22:
I've never used nanopb under C++, but it looks like all of the pb_*.h files already have their own `extern "C"` wrappers built in.  Adding your own will confuse that.  As a quick guess, I would think you only need to add protection around the include of tahu.h.

wes-johnson, 2020-08-16:
This code has been reworked thanks to @jbrzozoski. Please try out the latest code on the develop branch.

closed 2020-08-16 by wes-johnson