eclipse-tahu/tahu #81: Static code analysis reported few errors and warning

vanidesai, 2020-07-15:
989: Possible memory leak. Dynamic memory stored in 'metric.value.string_value' allocated through function
'decode_metric' at line 623 can be lost at line 628
/tahu/client_libraries/c/src/tahu.c:628 | decode_payload()
Code: MLK.MIGHT | Severity: Error (2) | State: New | Status: Analyze | Taxonomy: C and C++

989: Possible memory leak. Dynamic memory stored in 'metric.value.string_value' allocated through function
'decode_metric' at line 623 can be lost at line 628
/tahu/client_libraries/c/src/tahu.c:628 | decode_payload()
Code: MLK.MIGHT | Severity: Error (2) | State: New | Status: Analyze | Taxonomy: C and C++ 

#1002: Pointer 'dataset->columns' returned from call to function 'calloc' at line 779 may be NULL and may be dereferenced at
line 783.
/tahu/client_libraries/c/src/tahu.c:783 | init_dataset()
Code: NPD.FUNC.MIGHT | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1003: Pointer 'udt_template.parameters' returned from call to function 'calloc' at line 588 may be NULL and will be
dereferenced at line 589.
/tahu/client_libraries/c/test/test.c:589 | publish_device_birth()
Code: NPD.FUNC.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1004: Do not use unsafe buffer copy functions -- consider using a safe variant such as strcpy_s() instead.
/tahu/client_libraries/c/test/test.c:364 | publish_node_birth()
Code: MS.BANNED.COPY | Severity: Warning (3) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1005: Array 'PB_DECODERS' of size 10 may use index value(s) 0..15
/tahu/client_libraries/c/src/pb_decode.c:904 | pb_decode_noinit()
Code: ABV.STACK | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1008: Do not use unsafe buffer copy functions -- consider using a safe variant such as strcpy_s() instead.
/tahu/client_libraries/c/src/tahu.c:176 | decode_metric()
Code: MS.BANNED.COPY | Severity: Warning (3) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1016: Pointer 'propertyset->values' checked for NULL at line 301 will be dereferenced at line 316.
/tahu/client_libraries/c/src/tahu.c:316 | add_property_to_set()
Code: NPD.CHECK.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

1021: Do not use unsafe buffer copy functions -- consider using a safe variant such as strcpy_s() instead.
/tahu/client_libraries/c/src/tahu.c:425 | add_simple_metric()
Code: MS.BANNED.COPY | Severity: Warning (3) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

1023: Pointer 'payload->metrics' returned from call to function 'calloc' at line 410 may be NULL and will be dereferenced at
line 422. Also there is one similar error on line 424.
/tahu/client_libraries/c/src/tahu.c:422 | add_simple_metric()
Code: NPD.FUNC.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1024: Non-void function does not return value
/tahu/client_libraries/c/src/tahu.c:200 | decode_metric()
Code: FUNCRET.GEN | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned
#1025: Array 'PB_ENCODERS' of size 10 may use index value(s) 0..15. Also there is one similar error on line 347.
/tahu/client_libraries/c/src/pb_encode.c:343 | default_extension_encoder()
Code: ABV.STACK | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned
#1026: Pointer 'payload->metrics' checked for NULL at line 271 will be dereferenced at line 281.
/tahu/client_libraries/c/src/tahu.c:281 | add_metric_to_payload()
Code: NPD.CHECK.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1028: Do not use unsafe buffer copy functions -- consider using a safe variant such as strcpy_s() instead.
/tahu/client_libraries/c/src/pb_decode.c:797 | pb_field_set_to_default()
Code: MS.BANNED.COPY | Severity: Warning (3) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

1033: Do not use unsafe buffer copy functions -- consider using a safe variant such as strcpy_s() instead.
/tahu/client_libraries/c/test/test.c:578 | publish_device_birth()
Code: MS.BANNED.COPY | Severity: Warning (3) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

1033: Do not use unsafe buffer copy functions -- consider using a safe variant such as strcpy_s() instead.
/tahu/client_libraries/c/test/test.c:578 | publish_device_birth()
Code: MS.BANNED.COPY | Severity: Warning (3) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

#1036: Memory leak. Dynamic memory stored in 'inbound_payload.uuid' allocated through function 'decode_payload' at line
148 is lost at line 292
/tahu/client_libraries/c/test/test.c:292 | my_message_callback()
[... 49 lines cut ...]
/tahu/client_libraries/c/src/tahu.c:422 | add_simple_metric()
Code: NPD.CHECK.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

1066: Memory leak. Dynamic memory stored in 'prop_metric.name' allocated through function 'init_metric' at line 461 is lost
at line 525
/tahu/client_libraries/c/test/test.c:525 | publish_node_birth()
Code: MLK.MUST | Severity: Error (2) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned
067: Pointer 'payload->metrics' returned from call to function 'calloc' at line 270 may be NULL and will be dereferenced at
line 281.
/tahu/client_libraries/c/src/tahu.c:281 | add_metric_to_payload()
Code: NPD.FUNC.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned
067: Pointer 'payload->metrics' returned from call to function 'calloc' at line 270 may be NULL and will be dereferenced at
line 281.
/tahu/client_libraries/c/src/tahu.c:281 | add_metric_to_payload()
Code: NPD.FUNC.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned
067: Pointer 'payload->metrics' returned from call to function 'calloc' at line 270 may be NULL and will be dereferenced at
line 281.
/tahu/client_libraries/c/src/tahu.c:281 | add_metric_to_payload()
Code: NPD.FUNC.MUST | Severity: Critical (1) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned
1081: Function 'strcpy' is deprecated. Replace with more secure equivalent like 'strcpy_s', add missing logic, or re-architect.
/tahu/client_libraries/c/test/test.c:485 | publish_node_birth()
Code: SV.BANNED.REQUIRED.COPY | Severity: Review (4) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned
#1082: Do not use unsafe buffer copy functions -- consider using a safe variant such as strcpy_s() instead.
/tahu/client_libraries/c/src/tahu.c:784 | init_dataset()
Code: MS.BANNED.COPY | Severity: Warning (3) | State: New | Status: Analyze | Taxonomy: C and C++ | Owner: unowned

jbrzozoski, 2020-07-15:
I have a major overhaul of the c client in my fork which I am trying to wrap up to start a pull request on.

https://github.com/jbrzozoski/tahu/tree/c_client_rework

That should fix many of these issues.

wes-johnson, 2020-08-16:
This code has been reworked thanks to @jbrzozoski. Please try out the latest code on the develop branch.

closed 2020-08-16 by wes-johnson