eclipse-tahu/tahu #82: Finding size of actual payload while encoding using encode_payload()

shaila123456789, 2020-07-17:
Hello,

I am trying to find a way to get the size of a payload to encode (using encode_payload API). 
As per the API usage, I need to allocate space big enough to hold the entire payload in encode format. But I need to hard-code this value as of now. If I use sizeof() on the payload variable, it calculates the size of struct org_eclipse_tahu_protobuf_Payload (which is datatype of payload). 

How can I find the size of the actual payload with values at runtime ?

I am using code from https://github.com/jbrzozoski/tahu/tree/c_client_rework.

e.g. in test program, buffer_length = 128 has been hard-coded. I need to have this value calculated at runtime, as each time the payload will have different sizes.
                        org_eclipse_tahu_protobuf_Payload ddata_payload;
			// The binary_buffer must be large enough to hold the contents of the binary payload
			size_t buffer_length = 128;
			uint8_t *binary_buffer = (uint8_t *)malloc(buffer_length * sizeof(uint8_t));
			size_t message_length = encode_payload(binary_buffer, buffer_length, &ddata_payload);
			
-Thanks,
S

jbrzozoski, 2020-07-17:
There is a separate NanoPB function called `pb_get_encoded_size` to do this, but it isn't directly exposed through any helper functions yet.
This should be a workable example:
```org_eclipse_tahu_protobuf_Payload ddata_payload;
// The binary_buffer must be large enough to hold the contents of the binary payload
size_t buffer_length;
bool encode_passed = pb_get_encoded_size(&buffer_length, org_eclipse_tahu_protobuf_Payload_fields, &ddata_payload);
uint8_t *binary_buffer = (uint8_t *)malloc(buffer_length * sizeof(uint8_t));
size_t message_length = encode_payload(binary_buffer, buffer_length, &ddata_payload);
```

jbrzozoski, 2020-07-17:
I'm trying to get my branch updated and into a pull request soon.  I'll add a feature to find the size more intuitively.

shaila123456789, 2020-07-20:
Thanks for the reply  jbrzozoski !

This worked as expected. 
Apart from this, there is one more thing. I am getting lot of console logs from tahu library. Is there any way in which I can turn off these console logs ?

-Thanks,
S

jbrzozoski, 2020-07-20:
Near the top of tahu.h, change this define from 1 to 0:

```
// Enable/disable debug messages
#define SPARKPLUG_DEBUG 1
```

shaila123456789, 2020-07-24:
Hello,
I took the code from develop branch. I see you have merged earlier branch there. I set the SPARKPLUG_DEBUG to 0 and compiled the library. But still got the messages on console. (I had commented out all the fprintf's from test.c file). 
Commenting out the complete line #define SPARKPLUG_DEBUG 1 worked, as obviously compiler will not find this pre-processor.
But apart from this, I am looking for a command line parameter while compiling the library itself. I don't want to make changes in the code and then compile the library. Instead if I can add a parameter in make command that will be much helpful.

-thanks,
S

shaila123456789, 2020-09-15:
Hello jbrzozoski,

I wanted to stop tahu from showing console logs and I was looking for a command line parameter while compiling the library itself. Is any such compile time parameter available now to stop the console logs ?

-thanks,
S

jbrzozoski, 2020-09-28:
There is currently no compile time option.