eclipse-tahu/tahu #94: Errors in /tahu/client_libraries/javascript/sparkplug-payload/lib/sparkplugpayload.js

HelloHungryImDad, 2020-08-25:
Hi,

Apologies but I'm having issues trying to submit a pull request. In this file there are what appear to be a few errors which are preventing me from reusing this library. I might be wrong however!

These are: 
1. In getValue() and setValue() UInt32 should be in intValue, not longValue as this requires an unecessary conversion at the receiving end.
2. In decodeType() "DateTime" is missing.

Again, apologies for not going through the proper channels but I'm having issues with approval for pull requests.

Thanks,

Rikki

gaillarddamien, 2022-03-25:
Hi,
The fixes you provided here are perfectly legit.
I'm not sure what needs to be done to get these approved and merged?
I don't know whether I could/should do this on your behalf?

wes-johnson, 2022-03-25:
The issue was that the Github user account did not have a signed Eclipse Contributor Agreement attached to it (https://www.eclipse.org/legal/ECA.php). PRs can not be merged that have not agreed to the ECA. Once this is signed, and the PR is resubmitted, it can be merged.

closed 2022-03-25 by wes-johnson