eclipse-tahu/tahu #98: Typo in standard

masonpawsey, 2020-10-06:
I believe there's a typo in [the standard](https://www.eclipse.org/tahu/spec/Sparkplug%20Topic%20Namespace%20and%20State%20ManagementV2.2-with%20appendix%20B%20format%20-%20Eclipse.pdf); section 9.1 references section 7.4.1, which doesn't exist. I assume this is supposed to be section 7.3.1?

jbrzozoski, 2020-10-09:
It's definitely wrong, but I think it should actually refer to section 7.1.2 for the NBIRTH message.

masonpawsey, 2020-10-09:
Ah, I believe you're correct, good catch!

wes-johnson, 2022-01-25:
This spec is being rewritten as part of the Eclipse Sparkplug Working group. The Sparkplug Specification project Github repo is now here: https://github.com/eclipse/sparkplug. All new work is being done on the develop branch of that repo.

closed 2022-01-25 by wes-johnson