eclipse-tahu/tahu #104: master and develop fail to compile.

jmford-engr, 2020-11-02:
I just cloned the development branch of this project, but it seems the standalone C examples won't compile as they reference sparkplug_b.h.  The master branch won't compile either, since the library won't compile due to some leftover com_cirrus cruft.

The standalone example udt_example:
(base) maps@jftest:~/Documents/Projects/mirrorlab/casting/code/MQTT/tahu/sparkplug_b/stand_alone_examples/c/udt_example$ make
gcc -g -Wall -I../../../../client_libraries/c/include/ -c example.c -o example.o
example.c:17:10: fatal error: sparkplug_b.h: No such file or directory
   17 | #include <sparkplug_b.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:28: example.o] Error 1
(base) maps@jftest:~/Documents/Projects/mirrorlab/casting/code/MQTT/tahu/sparkplug_b/stand_alone_examples/c/udt_example$ 

So I tried changing sparkplug_b.h to tahu.h, but that's not helpful either:

[foo.txt](https://github.com/eclipse/tahu/files/5478468/foo.txt)

jbrzozoski, 2020-11-04:
Can you double check that your local copy is checked out to this commit:
```commit a3bf673bccb043293c3467ecad6c2b6bbd1a3533 (HEAD -> develop, origin/develop)
Merge: 8de1469 82617b5
Author: ibinshtok <ilya.binshtok@cirrus-link.com>
Date:   Tue Oct 20 10:22:29 2020 -0500

    Merge pull request #102 from ibinshtok/IMM-2102-A
    
    [IMM-2102] Fixed the 'ConvertMetrics' code in SparkplugBPayload Encod…
```
You can just run `git log -1` to get full info on your current checkout.
At that commit revision, there are no references to sparkplug_b.h and it builds fine for me.

jbrzozoski, 2020-11-04:
(Deleted my prior comment since I mis-read your question...)

jbrzozoski, 2020-11-04:
Agreed.  Those standalone samples are out of date.

jmford-engr, 2020-11-04:
would it be useful for me to fix it and submit it or will you do it?

jbrzozoski, 2020-11-04:
If you can handle it, I would appreciate that.  I'll review any pull request you submit.

jmford-engr, 2020-11-04:
OK.  I'll have a go at it.  I need to figure out how it all works anyway!

jmford-engr, 2020-11-05:
I've fixed all the problems in standalone_examples/c, but I'm having trouble with the pull request.  I forgot to sign the commit initially, but I fixed that.  I don't know how to fix the ECA signing problem.  I did fill out the ECA form.   Can you point me to some docs?  Obviously I am a github newbie.

jbrzozoski, 2021-01-20:
This was fixed by #107 and can be closed.

closed 2021-01-23 by wes-johnson