eclipse-tahu/tahu #217: Issue while trying to compile sparkplug_b.proto into Python code

mkashwin, 2022-08-22:
I was trying to compile the [sparkplug_b.proto](https://github.com/eclipse/tahu/blob/master/sparkplug_b/sparkplug_b.proto) file to generate the required python files for my project. 
The python file generated appears to be incomplete and not compiling. Would appreciate some guidance on what I might be going wrong here

**Command Used**:

> 
  ```
  protoc.exe  --python_out=./src/generated  - I ./sparkplug_b sparkplug_b.proto
  ```

**Environment Details - Windows 10**:
>
  ```ps
    protoc.exe  --version                                                     
    libprotoc 3.21.5
    
    python --version
    Python 3.10.5
  ```
Attached is the generated file:
[sparkplug_b_pb2.py.txt](https://github.com/eclipse/tahu/files/9394029/sparkplug_b_pb2.py.txt)

mkashwin, 2022-08-24:
Quick update 
I realized that the proto file is correctly being compiled by version [Protocol Buffers v3.19.4](https://github.com/protocolbuffers/protobuf/releases/tag/v3.19.4)

This issue has come up since [v3.20.0](https://github.com/protocolbuffers/protobuf/releases/tag/v3.20.0)
Recommend that the documentation is updated to reflect the dependency or the proto file structure is update to be compatible with the newer version of protobuf

tgrusendorf, 2023-01-08:
The file that you generated looks the same as what I've generated with protoc 3.21.6, and it is functional for me.  Is yours not working?

If you're wondering why it look so different from previous versions, see the following:
https://stackoverflow.com/questions/71960226/protocol-buffers-in-python-no-classes-generated
https://stackoverflow.com/questions/73932985/python-proto-is-not-generating-classes

closed 2023-01-10 by mkashwin