eclipse-tahu/tahu #286: Maven build fails

RickBullotta, 2023-04-18:
Latest version of Maven
JDK 18
Windows 10
Latest master code from Github

[ERROR] An internal error occurred
java.util.ConcurrentModificationException
    at java.util.TreeMap.callMappingFunctionWithCheck (TreeMap.java:750)
    at java.util.TreeMap.computeIfAbsent (TreeMap.java:604)
    at aQute.bnd.osgi.Jar.putResource (Jar.java:288)
    at aQute.bnd.osgi.Jar$1.visitFile (Jar.java:202)
    at aQute.bnd.osgi.Jar$1.visitFile (Jar.java:177)
    at java.nio.file.Files.walkFileTree (Files.java:2812)
    at aQute.bnd.osgi.Jar.buildFromDirectory (Jar.java:176)
    at aQute.bnd.osgi.Jar.<init> (Jar.java:119)
    at aQute.bnd.osgi.Jar.<init> (Jar.java:172)
    at org.apache.felix.bundleplugin.BundlePlugin.getOSGiBuilder (BundlePlugin.java:604)
    at org.apache.felix.bundleplugin.ManifestPlugin.getAnalyzer (ManifestPlugin.java:285)
    at org.apache.felix.bundleplugin.ManifestPlugin.execute (ManifestPlugin.java:111)
    at org.apache.felix.bundleplugin.BundlePlugin.execute (BundlePlugin.java:364)
...

wes-johnson, 2024-02-27:
Is this still an issue? Do you have full logs?

splatch, 2024-05-31:
Old releases of `maven-bundle-plugin` and/or bnd library used during the build are not compatible with Java above 11. You either need to bump bundle plugin version or downgrade JDK to 11.