eclipse-tahu/tahu #316: Mismatch between encoding and decoding DataSets using js

suneab, 2023-09-21:
When encoding DataSets using `encodeDataSet`, the `setValue` function is used which points to the "default" types which is `enum DataType` in `sparkplug_b.proto`
When decoding DataSets using `decodeDataSet` , the `getDataSetValue` function is used which points to the correct types which is `DataSetValue` in `sparkplug_b.proto`

This mismatch was introduced by this commit: https://github.com/eclipse/tahu/commit/59d162907ea569e80075eef2668da0ebffb1978d 
which fixed the decode function to the correct types.

A function to encode datasets correctly is missing

suneab, 2023-10-03:
@ferm10n is this something you would be interested in fixing as well?

ferm10n, 2023-12-19:
Hey @suneab sorry for taking forever to reply, I'm not really wired into github notifications (there's so much noise!) and only just now saw the mention.

I'm trying to recall how this stuff works... When `decodeDataSet` is called, it calls `getDataSetValue` which has no `case` for type `8` (UInt64). In the commit you linked that I authored, I wrote:
> they can't all share the metric getValue, because they can have different possible values.

For some reason I thought DataSets didn't accept all the same value types as a normal metric. But after looking through the spec I don't see that. So maybe the fix is to replace calls to `getDataSetValue` with `getValue`. That'd make sense because `encodeDataSet` calls `setValue`, and doesn't have a special fn to encode the value just for datasets.

I'll create a MR that removes `getDataSetValue` and `getTemplateParamValue`, and replace the calls with `getValue`.

commit 16e8bf01 (2024-02-27): fix: getting DataSet and Template values

- added cases for all  types that can be encoded
- these fns wrap `getValue` now to keep things dry
- closes #316
closed 2024-01-19 by wes-johnson