eclipse-tahu/tahu #352: SparkpluGUI

guiklimek, 2024-02-04:
## SparkpluGUI

I created an open source, dedicated light software to display Sparkplug messages. I would like to share it with you.

Repository: https://github.com/Ambre-io/sparkplugui

### Version

v1.0.0

wes-johnson, 2024-02-27:
Very cool. Any interest in dual-licencing with EPL 2.0 and adding it to the Tahu project?

guiklimek, 2024-05-11:
Hello @wes-johnson,

Thank you for your answer! Of course it should be nice to add it to the Tahu project! Tell me how can proceed.

I think another cool thing to do in the future is to improve the Sparkplug Go implementation so it could become the official 'Tahu Go Client'.

wes-johnson, 2024-05-22:
@guiklimek - First off, you need to agree to the Eclipse 'ECA' so you can contribute. See the FAQ here: https://www.eclipse.org/legal/ecafaq.php.

After that, you can create a PR against the develop branch of Tahu and it can be included. It must also include the license files and follow the Eclipse guidelines.

I like the idea of the Go impl too! That should at least conform to the basic directory layout of the other impls too.