eclipse-tahu/tahu #366: Python string unpacking broken again

rosstitmarsh, 2024-03-06:
If the array has any any 0 length strings at the start or the end, the strip function on line 129 in convert_from_packed_string_array removes them.

e.g.
```python
a = ["", "foo", "", "", "bar", "", ""]
convert_from_packed_string_array(convert_to_packed_string_array(a))
# returns ["foo", "", "", "bar"]
```

Replacing `strip` with `removesuffix` would work in python >=3.9 or removing the `strip` and putting the `[:-1]` back on the end.

Alain-Godo, 2024-03-08:
Hi @rosstitmarsh, recently I did a PR that can solve that issue ( and others ). I'm waiting for the review-> https://github.com/eclipse/tahu/pull/367

Alain-Godo, 2024-03-08:
[image]

rosstitmarsh, 2024-03-08:
Fixed in #367