eclipse-tahu/tahu #389: key multiPart missing

marcoratto, 2024-06-18:
I have a problem using the key "**MultiPart**" for the key **"Metadata".**
I wrote a java sample code:
[IssueMetaDataMultiPart.java.txt](https://github.com/user-attachments/files/15887285/IssueMetaDataMultiPart.java.txt)

This is the output written on the console:
`
MetaData [**isMultiPart=true**, contentType=text/plain, size=12, seq=0, fileName=fileName, fileType=fileType, md5=md5, description=config]

Metric [name=file, alias=1234, timestamp=1718715733989, dataType=String, isHistorical=null, isTransient=null, metaData=MetaData [**isMultiPart=true**, contentType=text/plain, size=12, seq=0, fileName=fileName, fileType=fileType, md5=md5, description=config], properties=null, value=Hello world!, isNull=false]
`

This is the message send to the Broker MQTT:

> 

{
  "timestamp": 1718715733989,
  "metrics": [
    {
      "name": "file",
      "alias": 1234,
      "timestamp": 1718715733989,
      "dataType": "String",
      **"metaData": {
        "contentType": "text/plain",
        "size": 12,
        "seq": 0,
        "fileName": "fileName",
        "fileType": "fileType",
        "md5": "md5",
        "description": "config"
      },**
      "value": "Hello world!"
    }
  ],
  "seq": 0
}

the key **isMultiPart** is missing.
I tried the same code with the Tahu Libraries 1.0.0 and 1.0.7 but it doesn't work.

Someone could help me ?

thanks in advance
Marco

wes-johnson, 2024-06-18:
What tool are you using to view the resulting message?

marcoratto, 2024-06-18:
I'm using mqttfx but I also wrote a Java class for encode and decode sparkplug messages and I have always the same issue.

wes-johnson, 2024-06-18:
I know the last version of MQTT.fx I used didn't properly display all portions of the Metric including the MetaData. Based on your code, I think you've set it up properly. What does your decoder code look like? What if you use the Tahu decoder?

marcoratto, 2024-08-01:
I solved the problem using the latest version of the library 1.0.7.